AnyLoc/AnyLoc #1: Question about Nardo-Air datasets

euncheolChoi, 2023-08-03:
Thanks for sharing your great work. This isn't a direct question about AnyLoc, but I'd like to know how you were able to access the Nardo-Air dataset you used for testing. 
I couldn't find an access link when I googled it, is there any chance you could shed some light on that?

Thanks in advance.

JayKarhade, 2023-08-04:
Hi @euncheolChoi , the Nardo-Air dataset was collected by members of our lab, and they haven't made it publicly available yet. That would be the reason why you cannot access any link.

euncheolChoi, 2023-08-04:
Oh, then do you plan to share this dataset as a public one in the future?

shockwaveHe, 2023-08-04:
Hi @euncheolChoi, me and another guy in the lab collect the dataset. The Nardo-Air dataset is primarily used for another project. Idealy it will be release when we decide to submit the paper for that project, probably ICRA2024. By the time I will inform you.

euncheolChoi, 2023-08-06:
Thank you, and I look forward to seeing it as a public dataset soon !

JayKarhade, 2023-08-06:
Closing this for now :)

closed 2023-08-06 by JayKarhade