AnyLoc/AnyLoc #4: demo for image retrieval

aTunass, 2023-08-25:
HI, thanks for you great work, but would you release the demo for image retrieval, I'm new to this area, and when I read your demos I couldn't imagine how this cluster would be used in image retrieval.

TheProjectsGuy, 2023-08-26:
Hi @aTunass,
I suppose you want to perform image retrieval using our AnyLoc method. I'm assuming you'll be using the AnyLoc-DINOv2-VLAD features for this. Here's how you can go about it

1. Select the domain for loading the VLAD cluster centers (vocabulary): You can use the [HuggingFace Space](https://huggingface.co/spaces/TheProjectsGuy/AnyLoc) for this. Upload some representative images in the tSNE app and see where the global descriptors fall in the tSNE plot.
2. Generate global descriptors: You can do this directly by using our [Colab Notebook](https://colab.research.google.com/github/AnyLoc/AnyLoc/blob/main/demo/anyloc_vlad_generate_colab.ipynb) (upload your images to Google Drive or directly to the Colab space).
3. Once you have the global descriptors, this becomes a similarity search algorithm (nearest neighbor retrieval). You can use the [get_top_k_recall](https://github.com/AnyLoc/AnyLoc/blob/main/utilities.py#L390) function if they're all tensors (uses FAISS), or make your own. The notebook saves them as numpy arrays (each descriptor in a file).

Of course, you can also setup and run our repo locally.

> I couldn't imagine how this cluster would be used in image retrieval.

We use VLAD aggregation to form global descriptors. VLAD has two stages:

1. Finding cluster centers: This is done by clustering the local patch descriptors from a "bunch of images" in a database.
2. Creating global descriptors: This is done by descriptor assignment (to cluster centers - based on distance) and concatenation of residuals.

You can find out more about it from [R Arandjelovic (2013) -  All About VLAD](https://www.cv-foundation.org/openaccess/content_cvpr_2013/html/Arandjelovic_All_About_VLAD_2013_CVPR_paper.html). In stage 1 above, we need to select the right (representative) database. 
We find that DINOv2 features work best with VLAD when we use domain-wise splits (like Urban, Indoor, and Aerial databases). For more information, you can read our paper at [N Keetha (2023) - AnyLoc: Towards Universal Visual Place Recognition](https://arxiv.org/abs/2308.00688), where we test this alongside many other aggregation methods (like GeM and global pooling).

Does this answer your doubt?

aTunass, 2023-08-27:
yeah, i see, i tried to use global descriptor and it worked. and one more question, if compared with BLIP in querying images by query image, will the result be better in the area of ​​place recognition right?

TheProjectsGuy, 2023-08-27:
I guess you're talking about the text-bonded image representations learned by BLIP in [J Li (2022) - BLIP: Bootstrapping Language-Image Pre-training for Unified Vision-Language Understanding and Generation](https://arxiv.org/abs/2201.12086). Do you mean the patch (local) descriptors (aggregated to form global descriptors) or CLS-like global descriptors for an image?
We did not try using BLIP (or the larger [LAVIS](https://github.com/salesforce/LAVIS) framework); this is because we found foundation models trained purely on images (like DINO and DINOv2) learn uniquely better representations than multi-modal models like CLIP (OpenAI's and OpenCLIP included).

aTunass, 2023-08-28:
Thank you very much, I will close this issue

closed 2023-08-28 by aTunass