AnyLoc/AnyLoc #9: How to achieve demo positioning effect？

wubalahao, 2023-09-07:
Hi I'm a newbie, how can I use anyloc to achieve the positioning effect in the demo demo locally? Specifically, given the left image, locate the position from the image library as below.
[image]

TheProjectsGuy, 2023-09-09:
Hey @wubalahao, I might have misunderstood your question

Do you want to achieve the same visual effects (UI) as our [interactive demo](https://anyloc.github.io/#interactive_demo) on our website https://anyloc.github.io/?
If yes, you can look at the website's codebase at https://github.com/AnyLoc/AnyLoc.github.io
Maybe you're looking for [this HTML](https://github.com/AnyLoc/AnyLoc.github.io/blob/main/index.html#L285) and [this JavaScript code](https://github.com/AnyLoc/AnyLoc.github.io/blob/main/demo/plot.js)

wubalahao, 2023-09-13:
> 嘿，我可能误解了你的问题
> 
> 您想实现与我们网站上 https://anyloc.github.io/ 上的[交互式演示](https://anyloc.github.io/#interactive_demo)相同的视觉效果 （UI） 吗？如果是，您可以查看网站的代码库 https://github.com/AnyLoc/AnyLoc.github.io 也许你正在寻找[这个 HTML](https://github.com/AnyLoc/AnyLoc.github.io/blob/main/index.html#L285) 和[这个 JavaScript 代码](https://github.com/AnyLoc/AnyLoc.github.io/blob/main/demo/plot.js)
To be more specific, how can I use them for image query after generating vlad files. Is there any readymade code or demo files?
[image]

wubalahao, 2023-09-14:
Well I wrote my own script to try the final image match, thanks to the author。

closed 2023-09-14 by wubalahao