AnyLoc/AnyLoc #13: VLAD Caching for the database and query

YznMur, 2023-10-17:
Hi,
I am tying to cache my own database and query, but faced this bug:

```
Unhandled exception
Traceback (most recent call last):
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/serialization.py", line 423, in save
    _save(obj, opened_zipfile, pickle_module, pickle_protocol)
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/serialization.py", line 650, in _save
    zip_file.write_record(name, storage.data_ptr(), num_bytes)
RuntimeError: [enforce fail at inline_container.cc:445] . PytorchStreamWriter failed writing file data/0: file write failed

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "dino_v2_global_vocab_vlad.py", line 737, in <module>
    main(largs)
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "dino_v2_global_vocab_vlad.py", line 528, in main
    db_vlads, qu_vlads = build_vlads_fm_global(largs, vpr_ds,
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "dino_v2_global_vocab_vlad.py", line 406, in build_vlads_fm_global
    db_vlads: torch.Tensor = vlad.generate_multi(full_db,
  File "/home/yazan/workspace/AnyLoc/utilities.py", line 1188, in generate_multi
    res = [self.generate(q, c) \
  File "/home/yazan/workspace/AnyLoc/utilities.py", line 1188, in <listcomp>
    res = [self.generate(q, c) \
  File "/home/yazan/workspace/AnyLoc/utilities.py", line 1107, in generate
    residuals = self.generate_res_vec(query_descs, cache_id)
  File "/home/yazan/workspace/AnyLoc/utilities.py", line 1242, in generate_res_vec
    torch.save(residuals,f"{self.cache_dir}/{cache_id}_r.pt")
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/serialization.py", line 424, in save
    return
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/serialization.py", line 290, in __exit__
    self.file_like.write_end_of_file()
RuntimeError: [enforce fail at inline_container.cc:325] . unexpected pos 320 vs 245 
```

Any ideas how to fix this?

YznMur, 2023-10-19:
Hi, @TheProjectsGuy, thanks for ur response
I'm running ```dino_v2_global_vocab_vlad``` with the default arguments
looks like I didn't have enough space for saving cashe, so I tested cashing a smaller dataset (170 images), but I got stuck into this problem:
```
Dino model loaded
Building VLAD cluster centers...
100%|█████████████████████████████████████████████████████████████████████████████████| 170/170 [00:02<00:00, 62.48it/s]
Database descriptors shape: torch.Size([170, 484, 384])
Descriptor dimensionality: 384
Caching cluster centers
VLAD cluster centers shape: torch.Size([8, 384]), (torch.float32)
Building VLADs for database...
Valid cache not found, doing forward pass
100%|█████████████████████████████████████████████████████████████████████████████████| 170/170 [00:03<00:00, 44.18it/s]
Database VLADs shape: torch.Size([170, 3072])
Building VLADs for query images...
Using cached VLADs for query images
Unhandled exception
Traceback (most recent call last):
  File "dino_v2_global_vocab_vlad.py", line 751, in <module>
    main(largs)
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "dino_v2_global_vocab_vlad.py", line 528, in main
    db_vlads, qu_vlads = build_vlads_fm_global(largs, vpr_ds,
  File "/home/yazan/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "dino_v2_global_vocab_vlad.py", line 421, in build_vlads_fm_global
    qu_vlads = vlad.generate_multi([None] * len(qu_img_names),
  File "/home/yazan/workspace/AnyLoc/utilities.py", line 1192, in generate_multi
    res = torch.stack(res)
RuntimeError: stack expects a non-empty TensorList
Program ended in 14.332 seconds
```

TheProjectsGuy, 2023-10-19:
It looks like you're trying to load from a cache (for query images). That's why it got this message

```
Using cached VLADs for query images
```
It originates from here
https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/dino_v2_global_vocab_vlad.py#L417-L421

But the cache (probably) doesn't contain valid data. Did you try the same program without caching?
If you don't, maybe check the cache folder for files. If you're using hard caching and the cache IDs are unique per sample/image, you should see label cache (the cluster each descriptor was assigned to) in the `{cache_id}_l.pt` files. This is cached here
https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/utilities.py#L841-L847

If you're using soft cluster assignment, the files should be ending with `_s.pt`
https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/utilities.py#L863-L868

Both should create residuals
https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/utilities.py#L951-L954

TheProjectsGuy, 2023-11-15:
Closing due to inactivity (and an earlier resolution). Feel free to open if any problems persist.

closed 2023-11-15 by TheProjectsGuy