AnyLoc/AnyLoc #15: need help when run  run the cmd python  dino_v2_vlad.py  on terminal

VClay, 2023-10-22:
there is an error 

print(db_vlads) in  script->dino_v2_vlad.py  row of  368th show tensor([NAN]) after run the cmd python  dino_v2_vlad.py  on terminal 

when I modify  script->dino_v2_vlad.py row of  93th  desc_facet: Literal["query", "key", "value", "token"] = "query"
     and modify anyloc (main directory)-> config.py row of 82th vg_dataset_name = "st_lucia"

the dataset download from the onedriver mentioned in the readme.md

tree ./st_lucia is
```
./st_lucia
|-- [  26]  images
|   `-- [  49]  test
|       |-- [148K]  database [1549 entries exceeds filelimit, not opening dir]
|       `-- [140K]  queries [1464 entries exceeds filelimit, not opening dir]
`-- [695K]  map_st_lucia.png
```
I would greatly appreciate it if you could reply

VClay, 2023-10-24:
It's strange that I didn't change anything and the problem disappeared. Currently, all the results are normal. I am very sorry for the inconvenience caused to you

TheProjectsGuy, 2023-10-25:
No problem. Closing the issue for now. Feel free to reply if you face any issues.

closed 2023-10-25 by TheProjectsGuy