AnyLoc/AnyLoc #16: Some questions that need help

VClay, 2023-10-30:
I have been reading this code recently. What I found is that the image input for extract_patch_descriptors is processed one by one. Would it be better to improve efficiency using batch size (if there are more GPUs)?

Another question is, instead of using dinov2 to train the benchmark's training dataset (e.g. Pitts30K/images/train) and then using dinov2 to test the benchmark's test dataset (e.g. Pitts310K/images/test), the AnyLoc project Just loaded the pretrained dinov2_vitg14 model to extract patch descriptors?

I would greatly appreciate it if you could reply

TheProjectsGuy, 2023-11-01:
Hey @VClay

Thanks for your observations

> What I found is that the image input for extract_patch_descriptors is processed one by one. Would it be better to improve efficiency using batch size (if there are more GPUs)?

I guess you're talking about this segment

https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py#L343-L361

Yes, you could use libraries like [joblib.Parallel](https://joblib.readthedocs.io/en/latest/generated/joblib.Parallel.html) to accelerate this. It uses about 6 GB VRAM, so if you have a GPU with say 32 GB VRAM, you could run five of these in parallel on the single GPU.

> the AnyLoc project Just loaded the pretrained dinov2_vitg14 model to extract patch descriptors?

I might not have understood you correctly here. What we do is create a [global dataset class](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py#L215) which has datasets of the same domain, and use it just for getting cluster centers

https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py#L369-L381

We then extract VLAD descriptors of the given dataset (the global dataset is not used after this)

https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py#L387

https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py#L411

TheProjectsGuy, 2023-11-01:
Also, if you're just testing our model, you can now try using [torch.hub](https://github.com/AnyLoc/AnyLoc/issues/11#issuecomment-1783870144)

We're working on restructuring the README to include this.

VClay, 2023-12-01:
I hope these can help improve the code

TheProjectsGuy, 2023-12-01:
Okay, I see what you mean. Thanks for the insight and proposing a DataParallel implementation for the codebase. We should keep this in mind for future releases. We're currently in the process of restructuring the repository to allow easier use.

VClay, 2023-12-02:
there are some things wrong in this code just for advice，