AnyLoc/AnyLoc #18: Recall  reproduce

babyporgrammer, 2023-11-04:
Hi thanks for your amazing work, the performance  is absolutely impressive. just one question, I am trying to reproduce the result of Recall rate, but find there is no code for it? any plan to release this part of code? your reply is very appreciate.

TheProjectsGuy, 2023-11-15:
Hey @babyporgrammer,

Thanks for taking an interest in our work. I might have misunderstood your question about the recall rate, but I'll try my best to answer.

> I am trying to reproduce the result of Recall rate, but find there is no code for it?

Though our [demos](https://github.com/AnyLoc/AnyLoc/tree/main/demo) do not have the code for reproducing recalls (because their purpose is only to do global feature extraction), we have the code for estimating recalls in our scripts like in [dino_v2_vlad](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_vlad.py#L371-L376) (which is our main AnyLoc-VLAD-DINOv2 method)

https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/dino_v2_vlad.py#L371-L376

The function `get_top_k_recall` can be found in [utilities.py](https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/utilities.py#L390-L394) of the main repository (not the demo folder).

Let me know if I was able to answer your query.

Nik-V9, 2023-12-08:
@TheProjectsGuy Maybe you should add a Benchmarking sub-section to the main README?

Nik-V9, 2024-04-04:
@TheProjectsGuy The scripts folder needs to be updated. It takes a lot of work to try out the best AnyLoc method.

Could you make a benchmarking folder with bash scripts for the best-performing variant of AnyLoc-DINOv2 (Different ViT architectures) and AnyLoc-DINO? 

Also, people can select the ViT architecture and dataset through args.