AnyLoc/AnyLoc #20: AttributeError: 'DinoV2ExtractFeatures' object has no attribute 'fh_handle'

angiebet, 2023-11-14:
custom data, when run :     python ./demo/anyloc_vlad_generate.py   raise 
Exception ignored in: <function DinoV2ExtractFeatures.__del__ at 0x7f8ce6d56b80>
Traceback (most recent call last):
  File "./AnyLoc/demo/utilities.py", line 101, in __del__
    self.fh_handle.remove()
AttributeError: 'DinoV2ExtractFeatures' object has no attribute 'fh_handle'

angiebet, 2023-11-14:
It's about Internet problem.I create an ipynb file ,test the module ,and it works.

		import torch
		torch.hub.list('facebookresearch/dinov2')
		model = torch.hub.load('facebookresearch/dinov2', 'dinov2_vitg14', pretrained=True)

extractor = DinoV2ExtractFeatures("dinov2_vitg14", desc_layer,
  File "./AnyLoc/demo/utilities.py", line 52, in __init__
    self.dino_model: nn.Module = torch.hub.load(
  File "/opt/conda/envs/anyloc/lib/python3.9/site-packages/torch/hub.py", line 539, in load
    repo_or_dir = _get_cache_or_reload(repo_or_dir, force_reload, trust_repo, "load",
  File "/opt/conda/envs/anyloc/lib/python3.9/site-packages/torch/hub.py", line 180, in _get_cache_or_reload
    repo_owner, repo_name, ref = _parse_repo_info(github)
  File "/opt/conda/envs/anyloc/lib/python3.9/site-packages/torch/hub.py", line 134, in _parse_repo_info
    with urlopen(f"https://github.com/{repo_owner}/{repo_name}/tree/main/"):
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/opt/conda/envs/anyloc/lib/python3.9/urllib/request.py", line 1350, in do_open
    r = h.getresponse()
  File "/opt/conda/envs/anyloc/lib/python3.9/http/client.py", line 1377, in getresponse
    response.begin()
  File "/opt/conda/envs/anyloc/lib/python3.9/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/opt/conda/envs/anyloc/lib/python3.9/http/client.py", line 289, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
Exception ignored in: <function DinoV2ExtractFeatures.__del__ at 0x7f37ee1e9b80>
Traceback (most recent call last):
  File "./AnyLoc/demo/utilities.py", line 101, in __del__
    self.fh_handle.remove()
AttributeError: 'DinoV2ExtractFeatures' object has no attribute 'fh_handle'

closed 2023-11-14 by angiebet

Wspiron, 2025-08-19:
> It's about Internet problem.I create an ipynb file ,test the module ,and it works.
>
> ```
> 	import torch
> 	torch.hub.list('facebookresearch/dinov2')
> 	model = torch.hub.load('facebookresearch/dinov2', [...]

Sorry to bother you, I would like to ask how to solve this problem, thx