AnyLoc/AnyLoc #24: NetVLAD Results

WHBSmith, 2024-02-04:
Hi,

Thank you so much for releasing your code for this interesting paper and documenting it. Quick question, you include code for MixVPR and CosPlace approaches that you used for comparison in your paper, but I can't find any references to NetVLAD. How did you produce the results for that approach, is there code for that in this repository?

TheProjectsGuy, 2024-02-04:
Hi @WHBSmith 

Thanks for your interest in our work. NetVLAD results in the benchmarks were produced using this script: https://github.com/AnyLoc/AnyLoc/blob/main/dvgl_benchmark/dvgl_many.sh

Let me know if this clarifies your doubts and if you have any other concerns.

WHBSmith, 2024-02-04:
Thanks, that's really helpful! Did you train your own model or did you use one of the pre-trained ones from the dvgl benchmark repo? There are quite a few different configurations!

TheProjectsGuy, 2024-02-05:
It's the `resnet18conv4` model that we found performs best across all baselines. We also tried the ViT NetVLAD model, but this one was better.

closed 2024-03-07 by WHBSmith