AnyLoc/AnyLoc #26: Question about the "Anyloc vocabularies"

euncheolChoi, 2024-02-26:
Thank you for sharing your great work.

I have some questions about  how to get "Anyloc vocabularies" which is different from "VLAD vocabularies".
According to the link below, the vocabulary by the Anyloc method is mentioned to be different from the VLAD vocabulary. 
([https://github.com/AnyLoc/AnyLoc/issues/22](url) )

I would like to reproduce the results of the paper, using the Vocabulary generated by the Anyloc method (I understood this to be the cluster center corresponding to each Domain) and Anyloc-VLAD-dinoV2.

To sum up,  my question is as belows.
1. How can i get vocabularies which is made by "Anyloc method"? 
- I'm wondering if I should download the vocabularies already created(e.g. download from links), or if I should do the PCA projection of descriptors by GeM global pooling as mentioned in the paper and get the vocabularies for each domain by myself.
If I need to create a vocabulary, can you tell me how to create?

2.  In Table 4 of the paper, the recall values are calculated for each dataset. For Anyloc-VLAD-DINOv2, I am curious to know which Vocabulary was used to obtain each of these results. In particular, I am interested in the results for the dataset in the Aerial domain. 
The reason I am interested in this is that I would like to reproduce the results in the paper. 
Additionally, to get the results of Anyloc-VLAD-DINOv2 after creating a cluster, can I use anyloc_vlad_generate.py?
[image]

Thank you.

TheProjectsGuy, 2024-03-07:
Hey @euncheolChoi 
Thanks for taking an interest in our work. I'll first review the vocabulary terminology, then answer your two questions.

## Vocabulary Terminology

Vocabulary is a "superset" of images we use to build database descriptors (for aggregation technique VLAD). By superset, I mean a collection of datasets. We experiment with the following vocabulary types:
- **Global**: where we include the database images from _all_ the datasets (in Table III and IV).
- **Structured**: where we include the database images from only the structured datasets (in Table III)
- **Unstructured**: where we include the database images from only the unstructured datasets (in Table IV)
- **Map-Specific**: where we include the database images from only the single dataset (on which we're testing). For example, if we're trying on Oxford, we "train" VLAD with only images from the Oxford database set (no other dataset). This can be read as "dataset-specific" as well.
- **Domain-Specific**: A "domain" is a collection of datasets with similar properties. We find this through tSNE and PCA projections (Figure 1, for example). We color-coded these domains in the paper. For example, the "Urban" domain comprises the Pitts-30k, St. Lucia, and Oxford datasets; the "Aerial" domain contains the Nardo Air datasets and the VP-Air dataset. In the paper, we project the GeM-pooled descriptors.

Unless specified otherwise, the results in Table III and IV are using the **domain-specific** vocabularies. We found these to give the best performance (see Table V).

This is further described in the sections III.D, V.B, and A2 of our [paper on arxiv](https://arxiv.org/abs/2308.00688v2).

## Getting Vocabularies

> How can i get vocabularies which is made by "Anyloc method"?

You could do it in either of the two methods. I suggest the first one.

### Method 1: Use the `torch.hub` model

This is still in beta but should suit your needs for benchmarking. You can refer to issue #11 for more on this. A simple tutorial is as follows

```py
# Load model
import torch
model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", 
        backbone="DINOv2", device="cuda")
# Images
img = torch.rand(1, 3, 224, 224)
# Result: VLAD descriptors of shape [1, 49152]
res = model(img)
# Also supports batching
img = torch.rand(16, 3, 224, 224)
# Result: VLAD descriptors of shape [16, 49152]
res = model(img)
# More help
print(torch.hub.list("AnyLoc/DINO"))
r = torch.hub.help("AnyLoc/DINO", "get_vlad_model")
print(r)
```

The default is the indoor domain (since we show major improvement there, and it's from the more widely available structured set). However, you can use the aerial domain by loading

```py
model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", 
        backbone="DINOv2", domain="aerial", device="cuda")
```

The above will load the model we used for the aerial dataset columns (Nardo Air and VP-Air) in Table IV.
This method works for our paper's indoor, urban, and aerial domains.

### Method 2: Using the repo's codebase

I think you should use this _only_ if you're going to replicate the results and not use this for benchmarking, etc. (because the repo setup is a little more tedious and requires setting up a container).

1. Set up the repository and the datasets as [described here](https://github.com/AnyLoc/AnyLoc/tree/f69501e0c9f80648483ac4e95c06b194642459a5?tab=readme-ov-file#validating-the-results)

2. Create the cluster centers using [dino_v2_global_vocab_vlad.py](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py). Call to the script is documented in [dino_v2_global_vocab_vlad_ablations.sh](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad_ablations.sh). This script creates a [GlobalVLADVocabularyDataset](https://github.com/AnyLoc/AnyLoc/blob/f69501e0c9f80648483ac4e95c06b194642459a5/scripts/dino_v2_global_vocab_vlad.py#L215) class and loads the datasets accordingly. Also note that you must run the scripts from the repo's home folder (instead of `cd`ing in the `./scripts` folder) - this is to find the other files and utilities in the repo. See this set of arguments to get a more accurate picture of which datasets are used for which vocabularies
https://github.com/AnyLoc/AnyLoc/blob/f69501e0c9f80648483ac4e95c06b194642459a5/scripts/dino_v2_global_vocab_vlad.py#L10-L58

3. When running the above script, you might want to "test" it on a small dataset (the bulk of the time will be spent on getting and caching cluster centers). You can later call with the same arguments to test on other datasets from the same vocabulary. The cluster centers will be loaded from the cache (if you're using it).

If you do not want to run step 2 (due to data or compute constraints), we also release the cluster centers in the [public material](https://iiitaphyd-my.sharepoint.com/:f:/g/personal/robotics_iiit_ac_in/EtpBLzBFfqdHljqQMnm6xdoBzW-4KFLXieXDVN4vPg84Lg?e=BP6ZW1). You'll have to download and unzip the `Colab1/cache.zip` file and navigate to `./cache/vocabulary/dinov2_vitg14/l31_value_c32` folder. You'll find a `c_centers.pt` file in each of the listed vocabulary folders. You could also get these from the [torch.hub release](https://github.com/AnyLoc/DINO/releases/tag/v1).

You can visualize the tSNE and PCA clusters by downloading datasets and using the scripts [dino_v2_datasets_gem_tsne_clustering.py](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_datasets_gem_tsne_clustering.py) and [dino_v2_datasets [... cut ...]
> Additionally, to get the results of Anyloc-VLAD-DINOv2 after creating a cluster, can I use anyloc_vlad_generate.py?

After you get the cluster centers (assuming you're following the method 2 above), you'll have to change the `c_centers_file` to this new `.pt` file here
https://github.com/AnyLoc/AnyLoc/blob/f69501e0c9f80648483ac4e95c06b194642459a5/demo/anyloc_vlad_generate.py#L143-L144

Assuming that you've configured the dataset directory, it should work fine. However, I recommend that you directly use method 1 as it doesn't require you to set up any repository for it to work.

euncheolChoi, 2024-03-14:
Thank you so much for the detailed guidelines. Of the methods you mentioned, 
**model = torch.hub.load("AnyLoc/DINO", "get_vlad_model",  backbone="DINOv2", domain="aerial", device="cuda")** 
to load the model and try to extract the descriptors. However, I am getting the following error while loading the model. Is this a temporary error or is there a solution to this, if you have one, I would appreciate it if you could share it.

------------ Dataset loaded ------------
------- Generating global descriptors -------
Using model : Anyloc-dino-VLAD_aerial_domain
Using cache found in /root/.cache/torch/hub/AnyLoc_DINO_main
Exception: invalid syntax (<fstring>, line 1)
[ERROR]: Exit is not safe
Traceback (most recent call last):
  File "/root/workspace/aerial_pr/Anyloc_docker_aerial/AnyLoc/aeria_scripts/dino_v2_global_vpr.py", line 327, in <module>
    main(largs)
  File "/opt/conda/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/root/workspace/aerial_pr/Anyloc_docker_aerial/AnyLoc/aeria_scripts/dino_v2_global_vpr.py", line 239, in main
    db_descs, qu_descs = build_cache(largs, gpr_ds)
  File "/opt/conda/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/root/workspace/aerial_pr/Anyloc_docker_aerial/AnyLoc/aeria_scripts/dino_v2_global_vpr.py", line 129, in build_cache
    anyloc = torch.hub.load("AnyLoc/DINO", "get_vlad_model", backbone="DINOv2", device="cuda")
  File "/opt/conda/lib/python3.7/site-packages/torch/hub.py", line 540, in load
    model = _load_local(repo_or_dir, model, *args, **kwargs)
  File "/opt/conda/lib/python3.7/site-packages/torch/hub.py", line 566, in _load_local
    hub_module = _import_module(MODULE_HUBCONF, hubconf_path)
  File "/opt/conda/lib/python3.7/site-packages/torch/hub.py", line 89, in _import_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 860, in get_code
  File "<frozen importlib._bootstrap_external>", line 791, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<fstring>", line 1
    (backbone = )

TheProjectsGuy, 2024-03-14:
Hey @euncheolChoi,

We didn't experience this error before. Are you trying this from a clean install? Make sure your torch hub directory - usually `~/.cache/torch/hub` - doesn't have anything before trying this out. Also ensure that you've sourced your conda environment with python and pytorch installed.

```
# Install dependencies
conda install -c conda-forge einops
pip install fast_pytorch_kmeans
```

And you can run the following

```py
import torch
# Load the model (this will download the vocabulary and the ViT model from torch.hub)
model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", backbone="DINOv2", domain="aerial", device="cuda")
# Your images here
img = torch.rand(16, 3, 224, 224)
# Global descriptors
res = model(img)    # (10, 49152) dim
```

Are you following the same steps as above?

I get the following when I run this
```
Python 3.10.13 | packaged by conda-forge | (main, Dec 23 2023, 15:36:39) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", backbone="DINOv2", domain="aerial", device="cuda")
Using cache found in /home/avneesh/.cache/torch/hub/AnyLoc_DINO_main
Storing (torch.hub) cache in: /home/avneesh/.cache/torch/hub/checkpoints/anyloc_files
100.0%
Downloading: "https://github.com/facebookresearch/dinov2/zipball/main" to /home/avneesh/.cache/torch/hub/main.zip
/home/avneesh/.cache/torch/hub/facebookresearch_dinov2_main/dinov2/layers/swiglu_ffn.py:51: UserWarning: xFormers is not available (SwiGLU)
  warnings.warn("xFormers is not available (SwiGLU)")
/home/avneesh/.cache/torch/hub/facebookresearch_dinov2_main/dinov2/layers/attention.py:33: UserWarning: xFormers is not available (Attention)
  warnings.warn("xFormers is not available (Attention)")
/home/avneesh/.cache/torch/hub/facebookresearch_dinov2_main/dinov2/layers/block.py:40: UserWarning: xFormers is not available (Block)
  warnings.warn("xFormers is not available (Block)")
Downloading: "https://dl.fbaipublicfiles.com/dinov2/dinov2_vitg14/dinov2_vitg14_pretrain.pth" to /home/avneesh/.cache/torch/hub/checkpoints/dinov2_vitg14_pretrain.pth
100.0%
VLAD caching is disabled.
Desc dim set to 1536
>>> img = torch.rand(16, 3, 224, 224)
>>> res = model(img)
>>> res.shape
torch.Size([16, 49152])
```

This takes about 5.3 GB on the GPU.

euncheolChoi, 2024-05-20:
Oh it was just my environment issue. Now i can get result. Thank you.

closed 2024-05-20 by euncheolChoi

JiahaoXia, 2024-10-30:
> Hey @euncheolChoi Thanks for taking an interest in our work. I'll first review the vocabulary terminology, then answer your two questions.
> 
> ## Vocabulary Terminology
> Vocabulary is a "superset" of images we use to build database descriptors (for aggregation technique VLAD). By superset, I mean a collection of datasets. We experiment with the following vocabulary types:
> 
> * **Global**: where we include the database images from _all_ the datasets (in Table III and IV).
> * **Structured**: where we include the database images from only the structured datasets (in Table III)
> * **Unstructured**: where we include the database images from only the unstructured datasets (in Table IV)
> * **Map-Specific**: where we include the database images from only the single dataset (on which we're testing). For example, if we're trying on Oxford, we "train" VLAD with only images from the Oxford database set (no other dataset). This can be read as "dataset-specific" as well.
> * **Domain-Specific**: A "domain" is a collection of datasets with similar properties. We find this through tSNE and PCA projections (Figure 1, for example). We color-coded these domains in the paper. For example, the "Urban" domain comprises the Pitts-30k, St. Lucia, and Oxford datasets; the "Aerial" domain contains the Nardo Air datasets and the VP-Air dataset. In the paper, we project the GeM-pooled descriptors.
> 
> Unless specified otherwise, the results in Table III and IV are using the **domain-specific** vocabularies. We found these to give the best performance (see Table V).
> 
> This is further described in the sections III.D, V.B, and A2 of our [paper on arxiv](https://arxiv.org/abs/2308.00688v2).
> 
> ## Getting Vocabularies
> > How can i get vocabularies which is made by "Anyloc method"?
> 
> You could do it in either of the two methods. I suggest the first one.
> 
> ### Method 1: Use the `torch.hub` model
> This is still in beta but should suit your needs for benchmarking. You can refer to issue #11 for more on this. A simple tutorial is as follows
> 
> ```python
> # Load model
> import torch
> model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", 
>         backbone="DINOv2", device="cuda")
> # Images
> img = torch.rand(1, 3, 224, 224)
> # Result: VLAD descriptors of shape [1, 49152]
> res = model(img)
> # Also supports batching
> img = torch.rand(16, 3, 224, 224)
> # Result: VLAD descriptors of shape [16, 49152]
> res = model(img)
> # More help
> print(torch.hub.list("AnyLoc/DINO"))
> r = torch.hub.help("AnyLoc/DINO", "get_vlad_model")
> print(r)
> ```
> 
> The default is the indoor domain (since we show major improvement there, and it's from the more widely available structured set). However, you can use the aerial domain by loading
> 
> ```python
> model = torch.hub.load("AnyLoc/DINO", "get_vlad_model", 
>         backbone="DINOv2", domain="aerial", device="cuda")
> ```
> 
> The above will load the model we used for the aerial dataset columns (Nardo Air and VP-Air) in Table IV. This method works for our paper's indoor, urban, and aerial domains.
> 
> ### Method 2: Using the repo's codebase
> I think you should use this _only_ if you're going to replicate the results and not use this for benchmarking, etc. (because the repo setup is a little more tedious and requires setting up a container).
> 
> 1. Set up the repository and the datasets as [described here](https://github.com/AnyLoc/AnyLoc/tree/f69501e0c9f80648483ac4e95c06b194642459a5?tab=readme-ov-file#validating-the-results)
> 2. Create the cluster centers using [dino_v2_global_vocab_vlad.py](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad.py). Call to the script is documented in [dino_v2_global_vocab_vlad_ablations.sh](https://github.com/AnyLoc/AnyLoc/blob/main/scripts/dino_v2_global_vocab_vlad_ablations.sh). This script creates a [GlobalVLADVocabularyDataset](https://github.com/AnyLoc/AnyLoc/blob/f69501e0c9f80648483ac4e95c06b194642459a5/scripts/dino_v2_global_vocab_vlad.py#L215) class and loads the datasets accordingly. Also note that you must run the scripts from the repo's home folder (instead of `cd`ing in the `./scripts` folder) - this is to find the other files and utilities in the repo. See this set of arguments to get a more accurate picture of which datasets are used for which vocabularies
>    https://github.com/AnyLoc/AnyLoc/blob/f69501e0c9f80648483ac4e95c06b194642459a5/scripts/dino_v2_global_vocab_vlad.py#L10-L58
> 3. When running the above script, you might want to "test" it on a small dataset (the bulk of the time will be spent on getting and caching cluster centers). You can later call with the same arguments to test on other datasets from the same vocabulary. The cluster centers will be loaded from the cache (if you're using it).
> 
> If you do not want to run step 2 (due to data or compute constraints), we also release the cluster centers in the [public material](https://iiitaphyd-my.sharepoint.com/:f:/g/personal/robotics_iiit_ac_in/EtpBLzBFfqdHljqQMnm6xdoBzW-4KFLXieXDVN4vPg84Lg?e=BP6ZW1). You'll have to download and unzip the `Colab1/cache.zip` file and navigate to `./cache/vocabulary/dinov2_vitg14/l31_value_c32` folder. You'll find a `c_centers.pt` file in each of the listed vocabulary folders. You could also get these from the [torch.hub release](https://github.com/AnyLoc/DINO/releases/tag/v1).
> 
[... cut ...]
Hi, @TheProjectsGuy. Thanks for your great work and kind explanations.
I have a question regarding the features used for generating VLAD descriptors.
According to your paper "Choice of Vocabulary" section, 
```Hence, we use vocabulary-independent global descriptors (DINOv2-GeM) and their (unsupervised) PCA projection to define vocabularies for VLAD aggregation.```

the dino_v2_global_vocab_vlad.py script only used DINOv2-GeM features to build VLAD cluster centers. Can you clarify if you used the PCA projection features to build the VLAD cluster centers?

TheProjectsGuy, 2024-11-02:
We use the GeM descriptor’s PCA projection to identify distinct regions in the (PCA projected) descriptor space. The idea is that these regions form a good representative sample. The vocabularies used for getting VLAD cluster centers are from the latent layers of DINO or DINOv2 (and not GeM). The VLAD scripts shouldn’t be using GeM descriptors. 
Also, we didn’t use any PCA projection _before_ doing VLAD. We propose PCA models where PCA is done _after_ VLAD aggregation. The `joint_pca_project.py` gives an overview.