AnyLoc/AnyLoc #28: Recall Reproduce on Baidu Mall dataset

abj247, 2024-04-05:
Thanks for sharing this great work!
While trying to reproduce the recall rates for the Baidu Mall dataset, I observed a drop in performance and couldn't match the recall rates mentioned in the paper
top@1= 43.1% (75.2%)
top@5 = 62.3% (87.6%)
The recall rates shown in the paper for AnyLoc-VLAD-DINOv2 are in parentheses. I am using the script from the `scripts` directory named `dino_v2_vlad.py`, using 32 clusters, 31 desc layer, dinov2_vitg14 architecture, key facet, and hard assignment.  I got satisfactory results on the St.Lucia dataset though using the same config.

Also tried using vocabulary from the cache directory and loading the VLAD cluster centers as given in the `demo` folder, tried using indoor vocabulary for dinov2_vitg14 with 32 clusters but the performance on the Baidu mall dataset decreased further. Not sure what's the problem causing this drop in performance, any ideas on how to fix this performance issue?

TheProjectsGuy, 2024-04-05:
Hey @abj247,

Thank you for your interest in our work. Did you use the indoor vocabulary or only the Baidu Mall database?

The `AnyLoc-VLAD-DINOv2` recall of 75.2% is using the [scripts/dino_v2_global_vocab_vlad.py](https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/dino_v2_global_vocab_vlad.py) file. You'll need the following database images for `indoor` domain.

https://github.com/AnyLoc/AnyLoc/blob/a9fda68c55083f765b019df148a1b67614f90ee2/scripts/dino_v2_global_vocab_vlad_ablations.sh#L105-L108

You seem to be using the [scripts/dino_v2_vlad.py](https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/dino_v2_vlad.py) file, which uses database images from only the Baidu Mall dataset.

abj247, 2024-04-06:
Thanks for the reply, I was only using the Baidu Mall database, I tried using [scripts/dino_v2_global_vocab_vlad.py](https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/dino_v2_global_vocab_vlad.py) with the indoor vocabulary containing database images from gardens, 17places, and baidu-datasets as you suggested but still I am not getting the mentioned recall rate, I got
R@1 = 47.07%
R@5= 64.96%
[image]
[image]

TheProjectsGuy, 2024-04-06:
Are you using the same NGC container and setup as described in the [Validating the Results](https://github.com/AnyLoc/AnyLoc#validating-the-results) section? Our run logs show a different number
[image]

These numbers are from the NGC container and the versions specified in the [setup_virtualenv_ngc.sh](https://github.com/AnyLoc/AnyLoc/blob/main/setup_virtualenv_ngc.sh) file.

abj247, 2024-04-06:
I was able to get similar results as shown in the paper for the Baidu mall dataset, just realized from the image you shared that I was using the key facet features instead of the value facet features.

Thanks for the help!