AnyLoc/AnyLoc #30: Nordland dataset

nobeldang, 2024-04-22:
Hi, I am unable to use the download_nordland.py since the link for downloading the dataset in the python file that hosts the nordland dataset is not working. Is the dataset hosted by AnyLoc available at different link? 

Error:

 raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='cloudstor.aarnet.edu.au', port=443): Max retries exceeded with url: /plus/s/8L7loyTZjK0FsWT/download?path=%2F&files=summer.tar.gz (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x0000021D03DC58B0>: Failed to establish a new connection: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond'))

oravus, 2024-04-24:
Hi @nobeldang, can you please try the following download links:
```
wget -O summer.tar.gz "https://universityofadelaide.app.box.com/index.php?rm=box_download_shared_file&shared_name=zkfk1akpbo5318fzqmtvlpp7030ex4up&file_id=f_1424421870101"

wget -O winter.tar.gz "https://universityofadelaide.app.box.com/index.php?rm=box_download_shared_file&shared_name=zkfk1akpbo5318fzqmtvlpp7030ex4up&file_id=f_1424424688104"

wget -O cleanImageNames.txt "https://universityofadelaide.app.box.com/index.php?rm=box_download_shared_file&shared_name=zkfk1akpbo5318fzqmtvlpp7030ex4up&file_id=f_1424408901067"

```

nobeldang, 2024-04-24:
Hi @oravus, thank you. This works.

Closing this issue,

closed 2024-04-24 by nobeldang