AnyLoc/AnyLoc #34: Missing datasets

bobhagen, 2024-06-06:
The public release link does not contain the datasets. Could you please share the dataset you used?

Best,
Robert

paolovic, 2024-06-12:
+ 1

TheProjectsGuy, 2024-06-12:
Hey there, there seems to be an issue with our IIIT-H Microsoft OneDrive. I'm in the process of sorting this out with the college. In the meantime, I have hosted all the public data on my personal Dropbox. I request you to download it from here: https://www.dropbox.com/scl/fi/dhyv4k932lxlbdygkolkf/AnyLoc2023-Public-Data.tar.gz?rlkey=8onlj7iezbiw0in1iul8breql&st=tjrn4vn9&dl=0

Please do not download it multiple times and use tools like `wget` or `aria2c` to download from command line (if you cannot keep a browser connection for a long time). The total download is about 32 GB and it contains all the datasets and cache.

This comment also addresses download issues caused by the onedrive downloader software (because the OneDrive link broke), possibly in issues #33 and #35

rosejn, 2024-08-23:
Thanks for making everything available to work with.  I'm not seeing the PCA parameters gzip file though.  Might you be able to share that one too so we don't need to regenerate descriptors for all the datasets?  I'm particularly interested in the global PCA.  Great work!

TheProjectsGuy, 2024-08-24:
Hey @rosejn,

Thanks for your interest in our work. We're working on releasing the PCA parameters too on [torch.hub](https://github.com/AnyLoc/DINO) (it's still experimental). In the meanwhile, you can achieve the PCA results using any of the scripts that save VLAD descriptors as cache and then using [joint_pca_project.py](https://github.com/AnyLoc/AnyLoc/blob/2ae462dbce05a2741fe8d075a1313488804e29a3/scripts/joint_pca_project.py) to obtain the projection. We'll include this too in the torch wrapper (by using the projection parameters) soon.

chennuo0125-HIT, 2025-03-24:
> Hey there, there seems to be an issue with our IIIT-H Microsoft OneDrive. I'm in the process of sorting this out with the college. In the meantime, I have hosted all the public data on my personal [...]

How to download with `wget` and what are the commands? I often stop downloading 4G when using web pages.

lpxlp, 2025-03-25:
> How to download with `wget` and what are the commands? I often stop downloading 4G when using web pages.

you can use powershell with 
Invoke-WebRequest -Uri "https://www.dropbox.com/scl/fi/dhyv4k932lxlbdygkolkf/AnyLoc2023-Public-Data.tar.gz?rlkey=8onlj7iezbiw0in1iul8breql&e=3&st=tjrn4vn9&dl=1" -OutFile "G:\VPR\AnyLoc2023-Public-Data.tar.gz" you can change your path I am downloading.

chennuo0125-HIT, 2025-03-26:
> you can use powershell with Invoke-WebRequest -Uri "https://www.dropbox.com/scl/fi/dhyv4k932lxlbdygkolkf/AnyLoc2023-Public-Data.tar.gz?rlkey=8onlj7iezbiw0in1iul8breql&e=3&st=tjrn4vn9&dl=1" -OutFile [...]

my os is ubuntu, how to download ? when i use `wget https://www.dropbox.com/scl/fi/dhyv4k932lxlbdygkolkf/AnyLoc2023-Public-Data.tar.gz?rlkey=8onlj7iezbiw0in1iul8breql&st=tjrn4vn9&dl=0`, cant download file.

lpxlp, 2025-03-26:
> 是这样的 我经过尝试 挂着梯子是下载不下来 经常在4-5G左右就会中断 我昨天回复你的时候我以为会成功 我让在国外的师兄下载好了 他晚上会上传到网盘 你要有需要联系我
>
> LP
>  @.***
>  …
>
> 怎么联系你？
>
>
> 是这样的 我经过尝试 挂着梯子是下载不下来 经常在4-5G左右就会中断 我昨天回复你的时候我以为会成功 我让在国外的师兄下载好了 他晚上会上传到网盘 你要有需要联系我
>
> LP
>  @.***
>  …
>
> 怎么联系你？

加我QQ?

LP
***@***.***

chennuo0125-HIT, 2025-03-26:
> 加我QQ?
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> LP
> ***@***.***
> […](#)

@lpxlp 可以直接发我邮箱 chennuo0125@163.com，麻烦了

chennuo0125-HIT, 2025-03-28:
> 是这样的 我经过尝试 挂着梯子是下载不下来 经常在4-5G左右就会中断 我昨天回复你的时候我以为会成功 我让在国外的师兄下载好了 他晚上会上传到网盘 你要有需要联系我
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> LP
> ***@***.***
> […](#)

@lpxlp 你好，请问可以共享以下网盘链接吗？

chennuo0125-HIT, 2025-04-27:
> Hey there, there seems to be an issue with our IIIT-H Microsoft OneDrive. I'm in the process of sorting this out with the college. In the meantime, I have hosted all the public data on my personal [...]

@TheProjectsGuy hello, thank for your work, Could all datasets be split into several small datasets? The current dataset is too large, and it will be truncated when I download to 10G.

hanyuzhubarry, 2025-09-26:
> 是这样的 我经过尝试 挂着梯子是下载不下来 经常在4-5G左右就会中断 我昨天回复你的时候我以为会成功 我让在国外的师兄下载好了 他晚上会上传到网盘 你要有需要联系我
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> LP
> ***@***.***
> […](#)

@lpxlp 我们正在进行相关的工作，想参考一下你们使用到的数据集，可以分享一下网盘链接吗，我的邮箱是3290296122@qq.com，万分感谢！