AnyLoc/AnyLoc #37: Are you training the VLAD vocabulary on the reference images?

amaralibey, 2024-07-04:
Hello dear authors,

Thank you for open sourcing the code.

I have read your paper and have a few questions. From my understanding, you are learning the vocabulary of VLAD from the reference images of each dataset you are testing on. Is that correct? Could you please clarify this aspect?

oravus, 2024-07-08:
Hi @amaralibey, the choice of data for constructing vocabulary is indeed one of the key aspects we explored, e.g, see Table 5 (vocabulary type) and Table 6 (vocabulary generalization).

Section 3D sets the premise for this (please also see [48] All about VLAD,  Arandjelovic & Zisserman, CVPR 2013, as your question is closely related to the use of map-specific vocabulary).

Hope the above answers your query.
Happy to provide more details otherwise, let me know. :)

TheProjectsGuy, 2024-07-08:
Hey @amaralibey,

You can also have a look at https://github.com/AnyLoc/AnyLoc/issues/26 for more information.

oravus, 2024-07-25:
Hope the above information addressed your queries, please feel free to reopen otherwise. Cheers!

closed 2024-07-25 by oravus