AnyLoc/AnyLoc #38: dataset avalible?

GyoungTaeChae, 2024-07-18:
hello, i want to try some codes in this repo.
however, when i click dataset link , it's all empty space 
how can i download datset? 

thankyou!

TheProjectsGuy, 2024-07-19:
Hey there, thanks for your interest in our work.
We're currently experiencing issues with Microsoft OneDrive. It's available on Dropbox through [here](https://www.dropbox.com/scl/fi/dhyv4k932lxlbdygkolkf/AnyLoc2023-Public-Data.tar.gz?rlkey=8onlj7iezbiw0in1iul8breql&st=tjrn4vn9&dl=0). The documentation and codebase will be updated with this soon.

Meanwhile, you can have a look at [documentation label](https://github.com/AnyLoc/AnyLoc/issues?q=label%3Adocumentation+) for more information on how to use the codebase.

GyoungTaeChae, 2024-07-19:
> Hey there, thanks for your interest in our work. We're currently experiencing issues with Microsoft OneDrive. It's available on Dropbox through [...]

really appreciate your reply, as i beginner of github, i couldn't check documentation label. Thank you !

oravus, 2024-07-25:
Hope you were able to download the dataset and use the codebase. Please feel free to reopen otherwise. Cheers!

closed 2024-07-25 by oravus